adamples/VU_meter #1: Screen refresh rate

capezoloh, 2018-03-15:
Neat little project you got here. Most of these kind of projects I see on youtube have really poor screen update rate. Is this true for yours aswell?

adamples, 2018-03-16:
Yeah, I also noticed it. There are several reasons for this. Most projects use u8g2, which is a great general-purpose library, but with some limitations arising from it's design goals. In the picture loop mode it refreshes the whole screen, which takes considerable amount of time just to transfer the data (~24ms for I2C at 400kHz). Additionally it uses synchronous I2C driver, so it can't render while sending the data either. This yields FPS in range of 20+.

In this project I aimed at different approach: asynchronous I2C, using HW available in atmega, and partial screen updates. It also uses sprites instead of drawing primitives directly to buffer. It all requires additional data and flash memory, but in the end you get update time of ~7ms, so more than 60FPS for 2 displays (it's more than display's actual frame rate I believe). It gives an impression of really smooth movement.

The code was tailored for purpose of displaying the needle against fixed background. It can be changed to do picture-loop style drawing or to use framebuffer, and calculating an update can be generalized too, but it's not the purpose of this project.

RaffRJ, 2018-05-05:
Hi Adam.
Its me raffy turc on youtube. 
I have downloaded and installed python 3.6 and compiled the 2 py files. Im getting error. What could be possibly the problem?

adamples, 2018-05-05:
@RaffRJ Can you please open new issue for this? Also, you have to provide at least some error message, describe what commands you executed. Otherwise it's really hard to tell.

closed 2018-05-05 by adamples