adamples/VU_meter #2: Compiling problems in windows

RaffRJ, 2018-05-05:
Using windows 64-bit machine

What I have done:
Copy the make.exe inside a folder with the SRC folder in it. When I envoke make, 1st problem  it was looking for missing file.

Now I have downloaded and installed python 3.6

Run the app and loaded python files. It returns error. Later will take screen shot of said error

adamples, 2018-05-06:
Ok, now I see the problem.

First of all you shouldn't copy `make` into repository. You just need to add it's directory to the PATH environment variable. Then you'll be able to run it from any directory. _Please remove it before proceeding with these instructions_.

Let's say you downloaded avr-gcc from here: http://blog.zakkemble.co.uk/avr-gcc-builds/ and unpacked to `C:\avr-gcc-7.3.0-x64-mingw\`. Then you open cmd, and add it to your PATH:

```
C:\vum> set PATH=C:\avr-gcc-7.3.0-x64-mingw\bin\;%PATH%
```

_Please double check if this is a valid path to bin directory, which contains `avr-gcc.exe` and `make.exe`_.

Now you can check if avr-gcc is recognized as a command:

```
C:\vum> avr-gcc --version
avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Same with make:

```
C:\vum> make --version
GNU Make 4.2.1
Built for Windows32
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

And Python:

```
C:\vum> python --version
Python 3.6.5
```

If all of that works, just execute `make` and you should get your `build\main.hex` file.

If you want to add avr-gcc to PATH permamently, use `setx` command:

```
C:\vum> setx PATH=C:\avr-gcc-7.3.0-x64-mingw\bin\;%PATH%
```

RaffRJ, 2018-05-07:
Update:

Hardware now working. Xtal changed to 20MHz. I have tried your older code and meter needle never moves. Tried all analog inputs.

And then I compiled again your latest code "using single lcd" . Now the meter shows pegged. Still no response from analog input pin/s (assuming this gets audio signal from A0)
[IMG]http://i68.tinypic.com/se7qq1.jpg[/IMG]

RaffRJ, 2018-05-07:
Got it!

So this is using Aref.no wonder 😀. My bad. Anyways at what voltage does the final op-amp output is with 0v input signal? Around 2.5v? Assuming using rail2rail output? I have here mcp602. Havent wired it yet according to your diagram.

adamples, 2018-05-07:
Yes, it will be somewhere around virtual ground, so 2.5V. If I remember correctly with input of sine wave with amplitude of ~0.5V it goes down to around 1.3V (I may be wrong about that, I tried many versions of this circuit).

In this code (main.c:93):
```
int16_t angle_l = (int32_t) (498 - adc_l) * 128 / 164;
```

498 is "zero" level, and 164 is scaling factor. You can adjust it to be more or less sensitive. Additionally you can change gain of input stage by changing the 47k resistor.

In my prototype I used mcp6002 which has both rail-to-rail inputs and outputs.

RaffRJ, 2018-05-07:
Copy that. So if I change 498 to 0, would this mean also as "referenced to GND"? 

and how do I turn off or disable Aref?

RaffRJ, 2018-05-09:
Adam

I have tried value of 800. The needle is in the "peg" position. And when I touch op-amp input, meter deflects to left. Quiet the opposite from your video. What couls be problem here? Its like the adc is configured for "reverse"

RaffRJ, 2018-05-09:
Adam.

Dis-regard my earlier post. Its working ok now. Ive scoped out the output of op-amp circuit. It registers virtual ground+sensed audio(positive voltage). It was the code that was inverting the results. Done some bit of google search(my mind kinda rusty with coding 😂) . Will rotate again 180° the bitmap image to "standard" orientation. Though I may add a bit of hardware mod at the adc input. Adjustable midpoint voltage then ac couple it to the op-amp/s front-end so I wouldnt be bothered adjusting code for "zero-ing""the meter needle.

Better if I could assign an un-used adc input to pot and map it (adding integer value) to the AN0 reading..but thats out of my head now. 
Will update a video later...

Could you suggest me an easier way of "zero-ing"  the needle on your code? 
Was thinking 8-bits(?) so 1024(or 1023) divide that in half since op-amp is referenced to 1/2 vcc. 
Then add maybe in increments of 5 ..you get my point.. lol.

I need some sleep 1st. Laters man!  And thanks for this vu project of yours.
Nice addition to my desktop eycandy oled display/s.

adamples, 2018-05-14:
Ok, I added quick fix to both main and single_display branches: https://github.com/adamples/VU_meter/commit/e31dd53ddfc5ed11ee5e7e64ef18e89d4039c249

Currently the code checks for needle angle to make peak indicator visible (main.c:106):

```
PEAK_INDICATOR_SPRITE.sprite.visible = (angle_l > 192);
```

closed 2018-05-23 by adamples