adamples/VU_meter #3: Request

classic-audio, 2018-06-11:
I really think you have a very fine project. I specially like the analog part and the very nice designed VU meter / display.
The very carefull designed analog part to design the correct VU meter simulation -thats very very good.

My request.
Would it be possible to get a copy of the software as an *.INO file?
This would make the project much more usefull for me.
I would be very happy to build the project.

adamples, 2018-06-13:
Hi, it's nice to hear that.

Do I understand correctly that your problem is in building the project and you would prefer to do it using Arduino IDE? Unfortunately it's not trivial to convert it, since some files are generated using Python scripts. But actually building it is not that hard, see this issue for detailed instruction: https://github.com/adamples/VU_meter/issues/2

If you need help with modifications we can work on that too.

classic-audio, 2018-06-15:
Hi,
Thank you for your reply.
Yes - I was hoping for a sketch for Arduino IDE, which is the platform I am able to handle. But I understand your point.
WHat I liked very much is your design of the meter graphics which is really beautiful and VU meter "like".
Next your very serius design of the analog electronics for the VU meter. It is not often projects are that prof.