adamples/VU_meter #5: The Circuit Diagram

classic-audio, 2019-04-14:
I am back and I am still thinking of building this great project.
To start - I would like to start with the circuit diagram /PCB.
I tried to open your eagle files in eagle 9.2.2  - but I am getting the message: This is not an eagle file.
I can open eagle files from eagle 6.x in eagle 9.2.2 so I expect also that eagle 8.x files should be possible to open but no.
Do you know a reason ?
Or would you be so kind to send the circuit in a PDF file?
Best regards
Frede Kristensen

adamples, 2019-04-15:
Did you try these files? (I put the wrong names in readme by mistake)
https://github.com/adamples/VU_meter/blob/master/hardware/vu_meter_2.sch
https://github.com/adamples/VU_meter/blob/master/hardware/vu_meter_2.brd

I will fix that and add schematic in pdf later today.

adamples, 2019-04-15:
I uploaded the circuit diagram here:
https://github.com/adamples/VU_meter/blob/master/hardware/vu_meter_sch.pdf

classic-audio, 2019-04-15:
Thank you very much. Really interresting for me to study you solution.
(The analog circuit).
I will have a look to night.