adamples/VU_meter #7: Display Question

DamianRaistrick, 2019-11-07:
Hi Adam
Really excellent project!
I'm attempting to get this to work on a couple of SH1106 displays with addresses 0x3c and 0x3d. I've managed to edit the config.h file to reflect the addresses and make the main.hex file (using terminal on my mac). I have managed to upload to an Arduino Uno using HexUploader but nothing appears on the displays.
I've tested the displays using the adafruit and U8Glib example code and they work fine on on I2C.
Is there anything else I should be adjusting before making the hex file for these displays?
Many thanks
Damian

adamples, 2019-11-08:
Hi! Yes, in fact you need to edit BOARD variable in Makefile: replace "BOARD?=WZ10" in 2nd line with "BOARD?=ARDUINO_UNO".

This project may not be of the same quality as u8glib, so if it still doesn't work, let me know ;)

DamianRaistrick, 2019-11-08:
Thanks so much Adam
Your project is inspiring! I only mentioned the other libraries/examples because I wanted to show I'd tested my displays were generally working ok with I2C.
I knew there was something I was doing wrong! :)
I'll try editing the Board variable as you suggest and let you know how I get on.
Thanks again
Damian

DamianRaistrick, 2019-11-08:
Hello again!
I edited the make file as instructed, but sadly I still have a blank display.
I wonder if trying a windows environment (rather than my mac) would make any difference?!
Incidentally the displays I'm using are these:
https://hobbycomponents.com/displays/621-3-sh1106-uoled-display-module-white
with the jumper set for I2C. DC pin sets the address to either 0x3C or 0x3D. CS tied to ground and RES is the reset pin normally tied to High.
Any other ideas?
Thanks
Damian

DamianRaistrick, 2019-11-08:
I've just tried an experiment to check HexUploader was flashing the Uno properly. I exported a hex file from the Arduino IDE of the Adafruit example code and flashed that using HexUploader and it worked fine.
So it looks like there's still something I'm doing wrong with the compiling of the hex file for your code.

DamianRaistrick, 2019-11-11:
Hi again Adam
OK, so now I have bought some different I2C displays (1306) and I've tried the code you have already complied to hex with my Uno (using your mega328p_16Mhz.hex) and things still don't work for me - blank displays again.
Going back to compiling myself - with my new 1306 I2C displays on 07xA and 07xB - what exactly should I be doing with the config.h file in terms of editing it and running it on my Arduino Uno? 
In your readme you say to edit the config.h file to "set proper OLED controller type". My guess is I'm not doing that properly. 
I see the list of controller type definitions in the config.h file, but the remark says not to edit these constants. So how should I edit the config.h file to set the controller type to 1306?
Many thanks
Damian

DamianRaistrick, 2019-11-11:
Don't worry! I now realise that these new displays do not have the addresses 0x7A and 0x78 when used with Arduino - they are 0x3C and 0x3D because of the way Arduino truncates the addresses.
The more I use Arduino, the more limited I realise it is!
I've returned the new displays and will now give up! I've wasted so much time trying to get this to work for me.
Nice project though! :-)
Feel free to close this 'issue'.
Best
Damian