adamples/VU_meter #8: PCB Production

ezhik1, 2020-03-15:
Hej Adam! Absolute stellar project! I too, find the ballistics and smoothing in analog vu meter very important :) My VU meters, one of my first [here](https://www.youtube.com/watch?v=RjcTkzzANwM), have all been based on lm3916 ICs and with a much larger footprint, so it would be real fun to play with this compact scale!

I saw most of your prototypes on youtube and looks like you've got the board produced already, did you have it made locally, or do you have something already set up with a factory in China? 

The BOM you have listed doesn't have designators, but I used Eagle to generate one from the .sch file, and it looks to be complete.

If I get the board produced, I want to make sure I've got the BOM right, Could you have a look?  Also, what was overall parts cost? I'm trying to see how economical this little visualizer will turn out.

Dziekuje! 

male pin   header: 1*2pin 2.54mm | 1 |   | CONN_02 | 1X02 | J2 | Multi connection point. Often   used as Generic Header-pin footprint for 0.1 inch spaced/style header   connections |   |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
male pin header: 2*3pin 2.54mm | 1 |   | CONN_03X22X3_SILK_MALE_PTH | 2X3 | J3 | Multi connection point. Often used as Generic Header-pin   footprint for 0.1 inch spaced/style header connections |   |   |   |   | CONN-09701 |   |  
male pin header: 2*4pin 2.54mm | 1 |   | CONN_04X2 | 2X4 | J4 | Multi connection point. Often used as Generic Header-pin   footprint for 0.1 inch spaced/style header connections |   |   |   |   |   |   |  
HC49SM-3.6864-30-50-60-16-ATF   ok? | 1 |   | CRYSTALHC49S | HC49/S | Q3 | CRYSTAL |   |   | 1667008 | unknown |   |   |  
100NF 50V 0805 | 10 | 100nF | C-EUC0805 | C0805 | C2, C6, C14, C18, C26, C27, C28, C30, C33, C34 | CAPACITOR, European symbol |   |   |   |   |   |   |  
Is TANT 100UF 6.3V B3528 part,   ok? | 2 | 100uF | CPOL-EUB/3528-21R | B/3528-21R | C29, C31 | POLARIZED CAPACITOR, European symbol |   |   |   |   |   |   |  
10k 1% 0805 | 15 | 10k | R-EU_R0805 | R0805 | R2, R4, R5, R8, R10, R12, R13, R16, R17, R19, R21, R23, R25,   R27, R37 | RESISTOR, European symbol |   |   |   |   |   |   |  
3296W-1-103LF | 2 | 10k | R-TRIMM3296W | RTRIM3296W | R6, R36 | Trimm resistor |   |   |   |   |   |   |  
10UF 16V 0805 | 1 | 10uF | C-EUC0805 | C0805 | C20 | CAPACITOR, European symbol |   |   |   |   |   | ceramic |  
Is TANT 10UF 16V B3528 ok? | 1 | 10uF | CPOL-EUB/3528-21R | B/3528-21R | C32 | POLARIZED CAPACITOR, European symbol |   |   |   |   |   |   |  
Plz confirm correct part is   CBC3225T100KR | 1 | 10uH | L-EUL3216C | L3216C | L1 | INDUCTOR, European symbol |   |   |   |   |   |   |  
Please confirm correct part is   LL4148 SOD80 | 2 | 1N4148 | DIODE-MINIMELF | MINIMELF | D2, D4 | DIODE |   |   |   |   |   |   |  
1k 1% 0805 | 2 | 1k | R-EU_R0805 | R0805 | R28, R33 | RESISTOR, European symbol |   |   |   |   |   |   |  
1UF 16V 0805 | 2 | 1uF | C-EUC0805 | C0805 | C5, C23 | CAPACITOR, European symbol |   |   |   |   |   | ceramic |  
220R | 2 | 220 | R-EU_R0805 | R0805 | R24, R32 | RESISTOR, European symbol |   |   |   |   |   |   |  
220k 1% 0805 | 4 | 220k | R-EU_M0805 | M0805 | R7, R9, R14, R30 | RESISTOR, European symbol |   |   |   |   |   |   |  
220k 1% 0805 | 6 | 220k | R-EU_R0805 | R0805 | R1, R11, R15, R18, R20, R34 | RESISTOR, European symbol |   |   |   |   |   |   |  
220NF 50V 0805 | 2 | 220nF | C-EUC0805 | C0805 | C4, C24 | CAPACITOR, European symbol |   |   |   |   |   |   |  
220PF 50V 0805 | 4 | 220pF | C-EUC0805 | C0805 | C7, C9, C17, C21 | CAPACITOR, European symbol |   |   |   |   |   |   |  
22k 1% 0805 | 2 | 22k | R-EU_R0805 | R0805 | R3, R35 | RESISTOR, European symbol |   |   |   |   |   |   |  
22PF 50V 0805 | 2 | 22pF | C-EUC0805 | C0805 | C1, C19 | CAPACITOR, European symbol |   |   |   |   |   |   |  
330NF 50V 0805 | 2 | 330n | C-EUC0805 | C0805 | C10, C13 | CAPACITOR, European symbol |   |   |   |   |   |   |  
330NF 50V 0805 | 2 | 330nF | C-EUC0805 | C0805 | C11, C12 | CAPACITOR, European symbol |   |   |   |   |   |   |  
33PF 50V 0805 | 2 | 33pF | C-EUC0805 | C0805 | C8, C25 | CAPACITOR, European symbol |   |   |   |   |   |   |  
S3B-PH-SM4-TB(LF)(SN) | 1 | 455-1750-1-ND | CONN_03 | 1X03 | J1 | Multi connection point. Often used as Generic Header-pin   footprint for 0.1 inch spaced/style header connections |   |   |   |   | XXX-00000 |   | 455-1750-1-ND
470NF 50V 0805 | 2 | 470nF | C-EUC0805 | C0805 | C3, C16 | CAPACITOR, European symbol |   |   |   |   |   |   |  
47k 1% 0805 | 2 | 47k | R-EU_M0805 | M0805 | R26, R31 | RESISTOR, European symbol |   |   |   |   |   |   |  
47NF 50V 0805 | 2 | 47nF | C-EUC0805 | C0805 | C15, C22 | CAPACITOR, European symbol |   |   |   |   |   |   |  
82k 1% 0805 | 2 | 82k | R-EU_R0805 | R0805 | R22, R29 | RESISTOR, European symbol |   |   |   |   |   |   |  
  | 2 | BAS70-04 | BAS70-04 | SOT23 | D1, D3 | Silicon Schottky Diodes |   |   |   |   |   |   |  
Please confirm correct part is   BAS85  SOD80 | 2 | BAT85 | DIODE-MINIMELF | MINIMELF | D5, D6 | DIODE |   |   |   |   |   |   |  
  | 2 | BSS138 | BSS123 | SOT23 | Q1, Q2 | N-CHANNEL MOS FET |   |   |   |   |   |   |  
MCP1525T-I/TT ok? | 1 | MCP1525 | MCP9700SMD | SOT23-3 | IC1 |   |   |   |   |   |   |   |  
MCP1700T-3302E/TT ok? | 1 | MCP1700 3V3 | MCP1703CB | SOT23 | IC6 | 250 mA, 16V, Low Quiescent Current LDO Regulator |   |   |   |   |   |   |  
We will supply MCP6004T-I/SL | 2 | MCP6004 | MCP604SL | SO14 | IC2, IC4 | Quad Op Amp 2.7V to 6.0V Single Supply CMOS |   |   |   |   |   |   |  
We will supply MCP6004T-I/SL | 1 | MCP604SL | MCP604SL | SO14 | IC3 | Quad Op Amp 2.7V to 6.0V Single Supply CMOS |   |   |   |   |   |   |  
ATMEGA88PA-AU ok? | 1 | MEGA88 | MEGA48/88/168-AU | TQFP32-08 | IC5 |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
REG1117  REG1117-ND     5-7work days | 1 | REG1117 | REG1117 | SOT223 | IC7 | 800mA and 1A Low Dropout (LDO) Positive Regulator |   | REG1117 | 1097566 | 14P6981 |   |   |

adamples, 2020-03-17:
Hi! I'm glad you like the project. I can look into this over the weekend, ok? But please consider this is just a hobby project, so maybe start by building it yourself, before putting it into mass production ;)

I ordered the PCBs in China but sourced parts locally and assembled the thing myself. So for me the cost was around $25 per board (including two displays and a lot of shipping costs). I didn't really care about money when designing this, as again, this was done just for fun, and not for profit.

ezhik1, 2020-03-17:
@adamples thanks for the insight! Yep, will definitely only go for hobby interest at this point. Would like to share with some friends/family, gifts and such :)

adamples, 2020-03-23:
HC49SM-3.6864 is 3.6MHz and SMD, and it should be 20MHz THT, like this one: https://eu.mouser.com/ProductDetail/Crystek-Corporation/CY22AS?qs=sGAEpiMZZMsBj6bBr9Q9aX0fcGvyQFE4zosMn%2FeWaHg%3D .

100uF caps (C29/C31) can be 6.3V, but I would suggest 10V or 16V. Same for C32: 16V is ok, but 25V is better.

CBC3225T100KR should be fine. You can update footprint in PCB to 3225 just to be sure.

LL4148 is fine. D5/D6 is any schottky diode with low reverse current, so BAS85 is ok (even 4148 would work here too).

MCP1525T-I/TT is ok. MCP1700T-3302E/TT is ok. ATMEGA88PA-AU is ok.

ezhik1, 2020-03-26:
Thank you for this!! agreed on the higher voltage capacitors, I've had plenty blow up on 12v circuits even when they're rated for 16v...just too close and cheap caps and poor tolerances go hand in hand :)

closed 2020-03-26 by ezhik1