adamples/VU_meter #9: compiling issues with Ubuntu

kuzmanzs, 2020-12-30:
Hi,

I want to compile your excellent VU meter project under Linux.
I have installed avr-gcc.
What should I need extra install to compile Arduino codes?

`avr-gcc -mmcu=atmega88p -DF_CPU="20000000UL" -MD -MP -Wall -Werror -std=c99 -ffunction-sections -fdata-sections -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -frename-registers -g -Wa,-ahlmsd=build/RELEASE/utils.lst -O2 -DNDEBUG   -c -o build/RELEASE/utils.o src/utils.c
In file included from src/utils.c:1:0:
src/utils.h:4:21: fatal error: avr/wdt.h: No such file or directory
compilation terminated.
`

make  --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu

avr-gcc --version
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.

adamples, 2020-12-30:
Hi, can you check if you have avr-libc and binutils-avr installed, and if so, what version?

kuzmanzs, 2021-01-03:
Hi,

binutils-avr is already the newest version (2.26.20160125+Atmel3.6.0-1)
for avr-libc i ahve installed http://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0.0.tar.bz2

thanks your help

kuzmanzs, 2021-01-03:
I have installed via apt avr-libc and it looks working, thanks.

closed 2021-01-03 by kuzmanzs